TiagoCortinhal/SalsaNext #6: ModuleNotFoundError: No module named 'tasks.semantic.modules.SalsaNextUncertainty'

chowkamlee81, 2020-07-01:
I started training by executing train.py along with command line arguments but i got error below

ModuleNotFoundError: No module named 'tasks.semantic.modules.SalsaNextUncertainty'

Kindly help

TiagoCortinhal, 2020-07-01:
Hey!

Again old class names relating to the Dropout. I tested and it seems fixed!

Let me know if something still doesn't work!

closed 2020-07-01 by TiagoCortinhal

chowkamlee81, 2020-07-01:
Still aim getting the smae error below as i copied your recent changes. Kindly help

  np_resource = np.dtype([("resource", np.ubyte, 1)])
Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 983, in _find_and_load
  File "<frozen importlib._bootstrap>", line 965, in _find_and_load_unlocked
ModuleNotFoundError: No module named 'tasks.semantic.modules.SalsaNextUncertainty'

chowkamlee81, 2020-07-01:
Sorry didn't copy exactly. Closing the issue