TiagoCortinhal/SalsaNext #10: GPU requirements to train the model

EthanZhangYi, 2020-07-17:
I use 8 TitanX Pascal GPUs to train the model following the README, and get the error below.
```
RuntimeError: CUDA out of memory. Tried to allocate 384.00 MiB (GPU 0; 11.91 GiB total capacity; 10.95 GiB already allocated; 250.56 MiB free; 11.15 GiB reserved in total by PyTorch)
```
Can you supply the hardware requirements? @TiagoCortinhal

TiagoCortinhal, 2020-07-17:
Hey @EthanZhangYi!!

We used 2 Quadro RTX 6000 24GB for most of our experiments. 

Can try decreasing the batch-size on [salsanext.yml](https://github.com/TiagoCortinhal/SalsaNext/blob/master/salsanext.yml)?

EthanZhangYi, 2020-07-23:
@TiagoCortinhal 
Thanks, the problem is solved.

closed 2020-07-23 by EthanZhangYi

finnSartoris, 2022-01-10:
When I try to train the model from scratch, my process gets killed after a few epochs due to memory issues. What is so memory consuming? I have already reduced the workers and the batch_size to 1. I am training on one NVIDIA GeForce GTX 1080 Ti

SSSSSSStar, 2023-05-05:
> When I try to train the model from scratch, my process gets killed after a few epochs due to memory issues. What is so memory consuming? I have already reduced the workers and the batch_size to 1. I am training on one NVIDIA GeForce GTX 1080 Ti

Have you solved the problem?

hahahaff, 2024-01-23:
@EthanZhangYi  Hello, how did you solve this problem？