TiagoCortinhal/SalsaNext #12: Bug: `proj_argmax` undefined.

EthanZhangYi, 2020-07-23:
https://github.com/TiagoCortinhal/SalsaNext/blob/1ee74804219635e660a50c6ed2f53dd09096db83/train/tasks/semantic/modules/user.py#L160-L169

TiagoCortinhal, 2020-07-23:
I just pushed it =)

closed 2020-07-23 by TiagoCortinhal

zeyu-hello, 2020-12-03:
@TiagoCortinhal 
You didn't fix it with `uncertainty` network.

TiagoCortinhal, 2020-12-03:
Hello @zeyu-hello!

If you check the if condition without uncertainty you will notice the only thing missing is `proj_argmax = proj_output[0].argmax(dim=0)`. I will push it later today (and check if anything else is missing) but for now you can try adding this!

Best

RaphaelLorenzo, 2021-07-23:
Hello @TiagoCortinhal and thank you for the great work ! It seems that `proj_argmax` it is still undefined for the case of `self.uncertainty==True`.
I fixed it by using : 
`proj_output2,log_var2 = self.model(proj_in)`
`proj_output = proj_output_r.var(dim=0, keepdim=True).mean(dim=1)`
` proj_argmax = proj_output2[0].argmax(dim=0)`

I Hope it is consistent ?

Using ` proj_argmax = proj_output[0].argmax(dim=0)` does not work (gives a 1 dimensional tensor of size 2048)