TiagoCortinhal/SalsaNext #13: val IoU of SalsaNext

EthanZhangYi, 2020-07-29:
@TiagoCortinhal ,
I train the SalsaNext and in the log folder three checkpoints are saved, `SalsaNext`, `SalsaNext_train_best` and `SalsaNext_valid_best`.
In `SalsaNext_valid_best`, the `best_val_iou` and `valid_iou` are 0.6101. However, I test `SalsaNext_valid_best` using `infer.py` and `evaluate_iou.py` , the average iou is 0.590, which is different from 0.6101.
Why are they different?

TiagoCortinhal, 2020-07-29:
Hey @EthanZhangYi !

That is a very good question. We barely adapt the eval setup from rangenet++. Nevertheless, I will investigate and come back to you once I have a clear answer!

shinya7y, 2020-08-25:
`trainer.py`: projected images' pixel-wise IoU
`infer.py` and `evaluate_iou.py`: point-wise IoU
The latter will be suitable for point clouds SS.

EthanZhangYi, 2020-08-25:
@shinya7y 
Thanks~

closed 2020-08-25 by EthanZhangYi