TiagoCortinhal/SalsaNext #16: What is data_cfg.yaml in infer.py

abcd40404, 2020-08-12:
I want to infer semantic-kitti dataset with pre-trained model.
When I run eval.sh, it required data_cfg.yaml. What information should I provide to run this code successfully?
BTW, I use salsanext.yml as arch_cfg.yaml, is that correct?

Thank you!

shinya7y, 2020-08-25:
Doesn't this work?
https://github.com/TiagoCortinhal/SalsaNext/blob/master/train/tasks/semantic/config/labels/semantic-kitti.yaml

TiagoCortinhal, 2020-08-31:
> Doesn't this work?
> https://github.com/TiagoCortinhal/SalsaNext/blob/master/train/tasks/semantic/config/labels/semantic-kitti.yaml

Hello @abcd40404, as @shinya7y said, data_cfg.yaml is the configuration for the dataset (labels, colours for the classes, train/valid/test splits, etc).

Arch_cfg.yaml is related to the arch which is salsanext.yml yes.

closed 2020-08-31 by TiagoCortinhal