TiagoCortinhal/SalsaNext #24: arch_cfg.yaml not found

SoftwareApe, 2020-10-15:
infer.py tries to load arch_cfg.yaml and data_cfg.yaml. Neither file can be found.

```python
    # open arch config file
    try:
        print("Opening arch config file from %s" % FLAGS.model)
        ARCH = yaml.safe_load(open(FLAGS.model + "/arch_cfg.yaml", 'r'))
    except Exception as e:
        print(e)
        print("Error opening arch yaml file.")
        quit()

    # open data config file
    try:
        print("Opening data config file from %s" % FLAGS.model)
        DATA = yaml.safe_load(open(FLAGS.model + "/data_cfg.yaml", 'r'))
    except Exception as e:
        print(e)
        print("Error opening data yaml file.")
        quit()
```

sebaleme, 2020-10-15:
We can retrain the model to generate them, but then providing the model alone is not enough

TiagoCortinhal, 2020-10-15:
@sebaleme I will upload them later today but you can use the ones generated by retraining (with standard options) as it will be the same.

sebaleme, 2020-10-15:
Hi, The arch_conf is in the repo, but not at the expected position: it is directly at the root level, with the following name: salsanext.yml

TiagoCortinhal, 2020-11-04:
@sebaleme, kNN should work in both types of projections, but for the scope of this paper we focused on SFV.

closed 2020-11-06 by TiagoCortinhal