TiagoCortinhal/SalsaNext #34: Question about the inference code in user.py

godspeed1989, 2020-11-12:
Thanks for your great work.
I had a question in inference code from user.py.
The order of model seems to be (mean, var)
But in user.py:
https://github.com/Halmstad-University/SalsaNext/blob/71114947a019c4760d52a4717cac531de12c67a2/train/tasks/semantic/modules/user.py#L157
And the `proj_output2` is used to get classification result:
https://github.com/Halmstad-University/SalsaNext/blob/71114947a019c4760d52a4717cac531de12c67a2/train/tasks/semantic/modules/user.py#L159

I am looking forward to your reply!

maxim0815, 2020-11-12:
I do have the same question. Within training you are doing it the right order:
```python
output_mean, output_var = adf.Softmax(dim=1, keep_variance_fn=keep_variance_fn)(*output)
```

TiagoCortinhal, 2020-11-12:
Hello @godspeed1989, @maxim0815!

You are correct it should be (mu,var). We tested with different possibilities to estimate the uncertainty and I must have forgotten to push the right order on the inference. 

I fix this issue, but I will have a deeper look as soon I have the time!

closed 2020-11-13 by godspeed1989