TiagoCortinhal/SalsaNext #37: bugs when train

cuge1995, 2020-11-27:
```shell
File "../../common/laserscan.py", line 190, in do_range_projection
    self.proj_range[proj_y, proj_x] = depth
IndexError: index -2147483648 is out of bounds for axis 0 with size 64
```

TiagoCortinhal, 2020-11-30:
Hey!

Could to provide more information about how this error occurs? (are you using the same dataset, etc) As I do not have this issue when I train.

closed 2020-12-09 by TiagoCortinhal

DesperateMaker, 2021-08-31:
I had this error. This is "-inf" appear when we devide by zero in the line 168:
        pitch = np.arcsin(scan_z / depth)

After this division by zero code will correctly to lines  ~214.
        proj_y = proj_y[order]
        proj_x = proj_x[order]
        
Projections will be correct, but after this you will have a lot of tensors with -inf