TiagoCortinhal/SalsaNext #38: SalsaNet to SalsaNext, difference from addition op to concat.

zeyu-hello, 2020-12-03:
Hi,
I read both SalsaNet and SalsaNext, and both are great work.
I noticed that you have change the skip connection in docoder from `add` to `concat`. Have you compared the performance between them, and what is the qualitative results.

Thanks.

TiagoCortinhal, 2020-12-03:
Hello @zeyu-hello!

We did compare, albeit we do not have an ablative study for that specifically. The main idea behind moving from moving from add to concat was motivated by PixelShuffle. Given that it will consume the channels for the upsampling, and trying to maintain a somewhat stable number of filters for it, without necessarily introducing new channels from other CNN layers.

TiagoCortinhal, 2020-12-03:
Hello2 @zeyu-hello!

We compared but we do not have any ablative study done on that. The main idea was to try to improve the ability of the network to merge information coming from the decoder side. This will help us maintain more representational information coming from the encoder, that will later be merged through block IV. 

I will try to find my runs with and without (I cannot assure I will find them) and if I find them I will share the differences! =)

zeyu-hello, 2020-12-04:
I know this `concat` is often applied in U-Net. The main idea I curious about this is that during deploymentation, `concat` op is not friendly for memory usage. If the error is somewhat acceptable, I could replace them to `add`.

Thanks in advance! @TiagoCortinhal

closed 2020-12-15 by zeyu-hello