TiagoCortinhal/SalsaNext #40: Set dropout to training in monte-carlo mode while inference

fcyeh, 2020-12-07:
https://github.com/Halmstad-University/SalsaNext/blob/cc8c75dc68d2607d16e2c82be61e7254e5b74a12/train/tasks/semantic/modules/user.py#L152

Should we set dropout to training when we do monte-carlo while inference?
Just like deep_uncertainty_estimation repo did: 
https://github.com/uzh-rpg/deep_uncertainty_estimation/blob/0093e12f234ad20da5a97aebb57ba060c8c4ca75/eval.py#L177
Thank you.

TiagoCortinhal, 2020-12-09:
Hello @fcyeh!

That could be a way of doing it, but as also stated on the paper we can use a pretrained model with their uncertainty estimation (which was our case).

fcyeh, 2020-12-09:
Hi @TiagoCortinhal. I mean if you don't do dropout in inference, the result will stay the same right? Then the variance will always be 0.

TiagoCortinhal, 2020-12-09:
Oh yes! I see what you mean. I lapse from my side. Same for the issue (both fixed). Thanks!

closed 2020-12-09 by TiagoCortinhal