TiagoCortinhal/SalsaNext #41: Some inference questions about final mean and variance

fcyeh, 2020-12-08:
[image]
Based on the paper A General Framework for Uncertainty Estimation in Deep Learning, I have questions about your inference code in user.py.

1. Should the final prediction be average of proj_output_r or proj_output2? The equation from the paper is average of T times monte-carlo predictions.
https://github.com/Halmstad-University/SalsaNext/blob/cc8c75dc68d2607d16e2c82be61e7254e5b74a12/train/tasks/semantic/modules/user.py#L157

2. Is log_var2 sensor uncertainty? If yes, based on the equation, we just need to do average. Why did you do log_var_r.var?
https://github.com/Halmstad-University/SalsaNext/blob/cc8c75dc68d2607d16e2c82be61e7254e5b74a12/train/tasks/semantic/modules/user.py#L159

Thank you.

TiagoCortinhal, 2020-12-09:
Hello @fcyeh!

It should be yes, but given our output is not a linear like in their case, taking directly the mean did not work.

fcyeh, 2020-12-09:
Hi @TiagoCortinhal. Thank you for the response. I think you were talking about question 1 right?
How about question 2, the sensor uncertainty? Why need to do var before average?
https://github.com/Halmstad-University/SalsaNext/blob/cc8c75dc68d2607d16e2c82be61e7254e5b74a12/train/tasks/semantic/modules/user.py#L159
[image]
Thank you.

TiagoCortinhal, 2020-12-09:
Yes it should be mean. This was corrected!

closed 2020-12-09 by TiagoCortinhal