TiagoCortinhal/SalsaNext #56: Pre-trained model Inference on CPU : Error during forward pass

Mipsology, 2021-04-02:
Hello,

- My aim to run inference without retraining
- I am running on Intel CPU  (i.e. no GPU)
- I am using the pre-trained model provided in this repo. 
- I got the complete datasets from SemanticKITTI. 
- I setup everything as per the instruction provided.  

When I run eval script with all variables, I get error in user.py on line 226 (which I believe is doing a forward pass on the model). 
The error says:  

> "Illegal instruction (core dumped)" 

Because of this, the Evaluation script fails when checking for length of pred_names (off course the inference never finished correctly)
I concluded this with some print statements in user.py and infer.py files. 

Any help or debug tips you can provide?   Thank You in advance! 

### Terminal Output : 

> (salanext) ~/zebra/NNs/SalsaNext/SalsaNext-master$ ./eval.sh -d dataset -p logs_preds -m saved_model -s train -c 30
> ----------
> INTERFACE:
> dataset /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/dataset
> log /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/logs_preds
> model /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/saved_model
> Uncertainty False
> Monte Carlo Sampling 30
> infering train
> ----------
> ----------
> Opening arch config file from /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/saved_model
> Opening data config file from /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/saved_model
> train 00
> train 01
> train 02
> train 03
> train 04
> train 05
> train 06
> train 07
> train 09
> train 10
> valid 08
> test 11
> test 12
> test 13
> test 14
> test 15
> test 16
> test 17
> test 18
> test 19
> test 20
> test 21
> model folder exists! Using model from /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/saved_model
> Sequences folder exists! Using sequences from /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/dataset/sequences
> parsing seq 00
> parsing seq 01
> parsing seq 02
> parsing seq 03
> parsing seq 04
> parsing seq 05
> parsing seq 06
> parsing seq 07
> parsing seq 09
> parsing seq 10
> Using 19130 scans from sequences [0, 1, 2, 3, 4, 5, 6, 7, 9, 10]
> Sequences folder exists! Using sequences from /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/dataset/sequences
> parsing seq 08
> Using 4071 scans from sequences [8]
> Sequences folder exists! Using sequences from /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/dataset/sequences
> parsing seq 11
> parsing seq 12
> parsing seq 13
> parsing seq 14
> parsing seq 15
> parsing seq 16
> parsing seq 17
> parsing seq 18
> parsing seq 19
> parsing seq 20
> parsing seq 21
> Using 20351 scans from sequences [11, 12, 13, 14, 15, 16, 17, 18, 19, 20, 21]
> ********************************************************************************
> Cleaning point-clouds with kNN post-processing
> kNN parameters:
> knn: 5
> search: 5
> sigma: 1.0
> cutoff: 1.0
> nclasses: 20
> ********************************************************************************
> Infering in device:  cpu
> *** JS INFO:  Entering infer ****
> *** JS INFO:  Entering infer_subset ****
> *** JS INFO: self.gpu =  False
> *** JS INFO:  Entering infer_subset NO uncertainty ****
> 
> **Illegal instruction (core dumped)**
> 
> finishing infering.
>  Starting evaluating
> ********************************************************************************
> INTERFACE:
> Data:  /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/dataset
> Predictions:  /home/mipso/zebra/NNs/SalsaNext/SalsaNext-master/logs_preds
> Split:  train
> Config:  config/labels/semantic-kitti.yaml
> Limit:  None
> ********************************************************************************
> Opening data config file config/labels/semantic-kitti.yaml
> Ignoring xentropy class  0  in IoU evaluation
> [IOU EVAL] IGNORE:  tensor([0])
> [IOU EVAL] INCLUDE:  tensor([ 1,  2,  3,  4,  5,  6,  7,  8,  9, 10, 11, 12, 13, 14, 15, 16, 17, 18,
>         19])
> Traceback (most recent call last):
>   File "./evaluate_iou.py", line 237, in <module>
>     eval(DATA["split"][FLAGS.split],splits,FLAGS.predictions)
>   File "./evaluate_iou.py", line 67, in eval
>     len(label_names) == len(pred_names))
> AssertionError
>

Mipsology, 2021-04-02:
One more detail.  I copied the following files in directory containing trained model : 

- salanext.yml  as  arch_cfg.yaml  
- semantic-kitti.yaml as data_cfg.yaml

Mipsology, 2021-04-02:
Hello, please ignore this issue. And if possible, please delete the issues to avoid any confusion for future visitors.  Apologies for any confusion caused.

closed 2021-04-02 by Mipsology

HuangVictorAuto, 2022-02-16:
@Mipsology , did you solve the problem? I encounter the same issue.