TiagoCortinhal/SalsaNext #58: Pretrained model : dead link

loukabvn, 2021-04-20:
Hello,
I'm a French student and I make an internship onto semantic segmentation, so I need to run your code, but the link to the pretrained model seems dead. If you have the time, can you update your link or send me another ?
Thank you !
(Sorry if it's not the place to ask that I'm not very familiar with Github)

TiagoCortinhal, 2021-04-20:
Salut!

I had an issue with my google drive. I will try to fix it as soon as possible!

loukabvn, 2021-04-20:
Thank you very much !

univjun, 2021-04-23:
Hi. TiagoCortinhal
Your results are impressive. 
I want to run your code. 
Could you fix the link to the pretrained model?

Thank you

TiagoCortinhal, 2021-04-23:
Thank you for reminding me!

I updated the link!

closed 2021-04-23 by TiagoCortinhal