TiagoCortinhal/SalsaNext #71: Errors occur during training.

LiXiang0021, 2022-04-01:
Thanks for author's remarkable working!

when i start to train this network without pretrained model, this error occurs showing below.
Can someone help me, thanks a lot!

/media/lijianguo/data_ssd/coding_test_platfrom/SalsaNext-master/train/tasks/semantic$ python3 train.py -d /media/lijianguo/data_ssd/kittidata/SemanticKITTI/data/dataset -ac /media/lijianguo/data_ssd/coding_test_platfrom/SalsaNext-master/salsanext.yml -l /media/lijianguo/data_ssd/coding_test_platfrom/SalsaNext-master/train_logs 
INTERFACE:
dataset /media/lijianguo/data_ssd/kittidata/SemanticKITTI/data/dataset
arch_cfg /media/lijianguo/data_ssd/coding_test_platfrom/SalsaNext-master/salsanext.yml
data_cfg config/labels/semantic-kitti.yaml
uncertainty False
Total of Trainable Parameters: 6.71M
log /media/lijianguo/data_ssd/coding_test_platfrom/SalsaNext-master/train_logs/logs/2022-4-01-18:46
pretrained None

Opening arch config file /media/lijianguo/data_ssd/coding_test_platfrom/SalsaNext-master/salsanext.yml
Opening data config file config/labels/semantic-kitti.yaml
Not creating new log file. Using pretrained directory
No pretrained directory found.
Copying files to None for further reference.
unsupported operand type(s) for +: 'NoneType' and 'str'
Error copying files, check permissions. Exiting...

AgentWizardX, 2022-11-07:
Good morning, were you able to solve this @LiXiang0021 ?

LiXiang0021, 2022-11-08:
Thanks for your reply and I had solved it.

AgentWizardX, 2022-11-08:
@LiXiang0021 How should I go about getting a log folder to be created or even recognize the pretrained directory?

TiagoCortinhal, 2022-11-08:
> @LiXiang0021 How should I go about getting a log folder to be created or even recognize the pretrained directory?

-l for the log -p for the pretrained

rohithsaro, 2022-11-09:
> Thanks for your reply and I had solved it.

@LiXiang0021 
Could you provide the solution ?

TiagoCortinhal, 2023-05-05:
Due to inactivity

closed 2023-05-05 by TiagoCortinhal