TiagoCortinhal/SalsaNext #74: Output Predictions and Input Point Clouds Mismatch

eyegotthis, 2023-04-11:
Hi there,

Thanks a ton for the amazing work, really appreciate. I was inferring the point cloud segmentations results from your script infer.py. I wanted segmentations for the training data for my own research but it seems that the output predictions do not match in size with the input point clouds. Tried the same for validations but this time there wasn't any mismatch. May I know what the reason could be for this and how I could obtain predictions that are equal in size to the input point clouds? Thanks a lot and looking forward to your response :)

Best,
Esas

TiagoCortinhal, 2023-04-11:
Hello!

Are you using SemanticKitti or another dataset?

TiagoCortinhal, 2023-04-12:
> Hi there!
> Thanks for the response! Im using SemanticKitti

Can you please share with me the error message?

eyegotthis, 2023-04-12:
There was not any error message thrown. The output predictions just had less number of points compared to the input point clouds when i opened the converted .label files back into numpy

eyegotthis, 2023-04-12:
I have visualized these output predictions from the semantickitti API and everything seems normal except that the far off points were sometimes missing.

TiagoCortinhal, 2023-04-13:
Did you turn off the Data augmentation when you do infer.py on the train slip? By detaul the training has DA which includes dropping points.

TiagoCortinhal, 2023-04-13:
You can change that here: https://github.com/TiagoCortinhal/SalsaNext/blob/a02fad97d646d4c132266ab79fbaea3ecfc237ed/train/tasks/semantic/dataset/kitti/parser.py#L326

eyegotthis, 2023-04-13:
Thank you very much! I think this was the reason, will try this out today and write back if everything has worked out.

closed 2023-04-13 by eyegotthis

juliangaal, 2023-04-17:
I am facing the same issue, but during validation set inference. Looking at your code, you are not dropping any points when loading the validation folder. Where else could the issue lie?