TiagoCortinhal/SalsaNext #79: The miou of the pretrained model on semantickitty is only about 58.97.

Xavier-wa, 2024-02-27:
Nice Work! But I have some problem about the miou of the pretrained model on semantickitty,when I upload the result of evalution,It is only 58.98 but not 59.5
[image]

kingskymoon, 2024-03-08:
my results on semantickitti seq 08 is 0.558?

Xavier-wa, 2024-03-12:
> my results on semantickitti seq 08 is 0.558?

you can modify the KNN["use"] to Ture and try again

Jaywxy, 2024-12-25:
my results on semantickitti seq 08 is  Acc avg 0.016 IoU avg 0.005   with parameters and` ./eval.sh -d /mnt/500_SSD/dataset/semantic-kitti/ -m pretrained -p preds -s valid -c 30`
# postproc parameters
post:
 CRF:
    use: False
    train: True
    params: False # this should be a dict when in use
  KNN:
    use: False # This parameter default is false
    params:
      knn: 5
      search: 5
      sigma: 1.0
      cutoff: 1.0